grigory-lobkov/rtsp-camera-view #1: Features request

Ziowebbo, 2018-04-03:
Hi! Interesting app, but we already talk about :)

I would like to request some features:

 - Possibility to send mail when video is lost
 - Possibility to put the program in FULL Screen without the bar and border
 - Possibility to resize the Video in the matrix to create (if user need it) a camera view in different dimension. This will allow to have a focus on a camera and have others smaller.
Try to add an image for example.

Thanks
[image]

grigory-lobkov, 2018-04-04:
Now I rewriting a source code to MVP model to have better scalebility. I hope, will finish it in April. Will try to do a fullscreen in May. It would be better, if you split your request next time for three separate features :)

Ziowebbo, 2018-04-04:
Ok mate!!! Thanks!!!

Inviato da Iphone

grigory-lobkov, 2018-04-26:
Fullscreen realized! [Download Beta Portable](https://github.com/grigory-lobkov/rtsp-camera-view/releases/tag/v1.0.0.0b)

Ziowebbo, 2018-04-27:
That's great Grigoy, thanks!!!!!!!!!!! :) :) :) :)

grigory-lobkov, 2018-05-05:
Have you tried 1.0.0.1 ?
Are  you using it, or wait for a new features ?

Ziowebbo, 2018-05-05:
Hi Grigory,
No tried the 1.0.0.1, I can’t find it on GitHub.
I tried the 1.0.0.0b and it works but I managed to do “full screen” only manually editing the xml.

And I have a question: there is a tag in the xml called “screen”. I thought it was useful to open the app directly in full screen on a certain screen if I have multi monitor, but I don’t know how to use it.
Is for this kind of use? 

Inviato da Iphone

grigory-lobkov, 2018-05-05:
Fullscreen fixed in 1.0.0.2
This options is for command line. Just add "screen=N", where N is number of monitor, started with 0.
You can also add "fullscreen" or "maximize" there.

You can find fresh beta in the same release:
https://github.com/grigory-lobkov/rtsp-camera-view/releases/tag/v1.0.0.0b

I'm planning to start new beta, when something really new will appear.
P.S. "full screen" bug was another "Issue" ;)  Next time, make it, please :) - it will be easier to search for somebody...

Ziowebbo, 2018-05-05:
Ok Grigory, thank you!

Inviato da Iphone

grigory-lobkov, 2018-05-06:
"fullscreen" available on right click 
[image]
"screen=N" from command line and settings.xml

grigory-lobkov, 2018-05-06:
:)
[image]

grigory-lobkov, 2018-05-06:
I've found a lot of bugs there, while testing. So, new betas and implements are coming soon... :)

Ziowebbo, 2018-05-07:
Grigory: installed the latest beta and is working very good! Need to
recreate the XML because tells me there was an error, but opened and worked
good.
What is not working is the screen number extension. It opens every time on
the screen where I launch the command.

Apart from that, is working very good and no more problem with the text on
the screen that disappear :)

Thanks again for your patience and support.

grigory-lobkov, 2018-05-07:
> tells me there was an error, but opened and worked good.

Does program take your settings, or you needed to recreate sources ?

> What is not working is the screen number extension. It opens every time on
the screen where I launch the command

Can you explain, how are you launching the app, maybe command line ?
Btw, screen number starts from zero in program ... (maybe it's habitually to increment it by one?)

> no more problem with the text on the screen that disappear

Haven't heard yet. Don't explain, cause it's good now)

Thank you for your suport and feedback)

Ziowebbo, 2018-05-09:
Hi Grigory, sorry for the delay:

tells me there was an error, but opened and worked good.

Does program take your settings, or you needed to recreate sources ?

                        The program take the settings, but tells me there
was an error on the XML. I saved the old XML with another name, launched
the program and let him save a new XML. So I opened the new XML and pasted
the <cams> tags from the old to new and worked without recreate the
sources. I can try to recover the old XML and send to you with the new so
is possibile to see differences.

What is not working is the screen number extension. It opens every time on
the screen where I launch the command

Can you explain, how are you launching the app, maybe command line ?
Btw, screen number starts from zero in program ... (maybe it's habitually
to increment it by one?)

                         I launched a cmd shell and write directly the
command with the screen=1. It opens everytime on the screen where the cmd
shell is positioned. For the moment, I created 3 different folders with the
program inside, with 3 different xmls and also created 3 windows shortcuts
pointing to each directory executable, placed them on each display and
launched from there I can put the program on the screen where I launch the
shotcut. Saving in the XML the fullscreen=1, I have the program going
fullscreen on the desktop where the shortcut is placed. On the shortcut I
didn't put any attributes, just start the program.
Hope I was clear :-)

no more problem with the text on the screen that disappear

Haven't heard yet. Don't explain, cause it's good now)

Thank you for your support and feedback)

                       Mate, thanks go to you for your patience, kindness
and availability :-)

Ziowebbo, 2018-05-13:
> support old settings.xml - fixed.
> some command line arguments skipped - fixed.
>
> Please, check 1.0.0.3 out! 1.0.0 BETAs
>
> Thanks for you help.

Great Grigory!
Tomorrow I’ll try the new version and let’s see if is the “screen=“ argument will work :-) 

Have a nice Sunday!

Inviato da Iphone

Ziowebbo, 2018-05-14:
Ok Grigory, I tried the new version. Now the "screen" option works, via
command line and also in the XML, but the app crashes whenever it starts.
It opens on the screen selected but crashes with no messages, windows just
says "program stopped". Click OK on the pop-up and need to reopen the app.

2018-05-13 10:57 GMT+02:00 Zio Webbo <ziowebbo@gmail.com>:

grigory-lobkov, 2018-05-14:
Can you provide a screen-shot?
What OS are you using? ... I tried WinXPSP2, Win10, cannot reproduce your error.
What Framework version?
Try to delete settings.cfg...
Try to take dll's from previous version...
Try to run without any command line parameters. I tried also:
"C:\Documents and Settings\Administrator\Desktop\rtsp-camera-view 1.0.0.3 beta\RtspCameraView.exe" screen=0 fullscreen

grigory-lobkov, 2018-05-15:
VLC 2.1.5 fresh installed.
1) WinXP SP3
Microsoft .NET Framework 4 Client Profile
Microsoft .NET Framework 4 Extended
You can download framework [here](https://www.microsoft.com/en-US/download/details.aspx?id=17718)
2) Windows 10.0.14393
No any additional framework

Everything worked fine.
What difference in our systems we have?

Ziowebbo, 2018-05-15:
Hi mate!
The system is:

Windows 7 SP1
Microsoft .NET Framework 4.7
VLC 2.1.5 fresh installed

Inviato da Iphone

grigory-lobkov, 2018-05-15:
Testing OS: Windows 7 Enterprise SP1
> C:\\> ver
> Microsoft Windows [Version 6.1.7601]

VLC 2.1.5 fresh installed

1. Microsoft .NET Framework 4.5
worked fine

2. Microsoft .NET Framework 4.7 NDP47-KB3186497-x86-x64-AllOS-ENU
worked fine

Please, check files rights, download again, reinstall Framework / VLC, try another OS / another beta.

Ziowebbo, 2018-05-17:
Hi mate, I made other test and this is what I understood: the program crashes at start sometimes, but systematically when you point to a screen that was different than the last start.

This problem is present only on this version.

grigory-lobkov, 2018-05-22:
Thank you for testing.

I tried to add margins on screen size, please, try [1.0.0.4 beta](https://github.com/grigory-lobkov/rtsp-camera-view/releases/tag/v1.0.0.0b) with just one parameter `screen=0` or `screen=1`

If it won't help, we will try another solution!

Ziowebbo, 2018-05-23:
Hi Grigory, unfortunately nothing changed.
At first it seems that worked well, but after some trying of closing and
re-opening, it started to crash.
I think is a problem of my configuration and honestly I tell you to do not
put much effort on this problem at the moment, unless you have identified a
possible cause.
Potentially is only MY problem and changing video drivers or other could
fix the problem, but I haven't found time to do this test.

:-)

grigory-lobkov, 2018-05-23:
I can't try multi-monitor configuration :(
I will try another solution in next beta(it's commented for now).

What version are you using as a production?

grigory-lobkov, 2018-06-18:
Please, check out crashes and email alerts in new [beta 1.0.0.5](https://github.com/grigory-lobkov/rtsp-camera-view/releases/tag/1.0.0.5)
... please, open a new Issue, if any ...

closed 2018-06-18 by grigory-lobkov