grigory-lobkov/rtsp-camera-view #3: Public RTSP sources as example

grigory-lobkov, 2018-05-15:
Since 10.05.2018 added some public RTSP sources as examples
- Hessdalen
http://freja.hiof.no:1935/rtplive/_definst_/hessdalen03.stream/playlist.m3u8
- USA golf
http://wmccpinetop.axiscam.net/mjpg/video.mjpg
- Highway
rtsp://170.93.143.139/rtplive/470011e600ef003a004ee33696235daa
- Semerkand TV
rtmp://semerkandglb.mediatriple.net:1935/semerkandliveedge/semerkand2

I'd like the source to have good and bad resolution sources to show program capabilities.

**Do you have any offers?**

hblanken, 2019-07-03:
rtsp://freja.hiof.no:1935/rtplive/_definst_/hessdalen03.stream

abhijeet2431, 2020-03-28:
https://g33ktricks.blogspot.com/p/the-rtsp-real-time-streaming-protocol.html
This one might help too.

grigory-lobkov, 2020-04-25:
I have found a good [source](https://www.insecam.org/en/byrating/) of public cameras.

Since 25.04.2020 the example cameras are
- Tokyo
http://210.148.114.53/-wvhttp-01-/GetOneShot?image_size=640x480&frame_count=1000000000

- Krasnodar, Sochi
http://158.58.130.148:80/mjpg/video.mjpg

- Antwerpen, Antwerpen
http://81.83.10.9:8001/mjpg/video.mjpg

- Colorado, Glenwood Springs
http://208.72.70.171:80/mjpg/video.mjpg

tabishimran, 2020-11-23:
security cams
http://46.16.226.181:88/?action=stream
http://79.132.232.174:8080/?action=stream
http://46.151.101.158:8081/?action=stream
http://90.66.111.157:8080/?action=stream
http://91.49.234.155:9000/?action=stream
http://46.151.101.149:8081/?action=stream
http://78.99.31.149:8081/?action=stream
http://46.151.101.134:8082/?action=stream
equipment cams
http://68.3.132.170:8080/?action=stream
http://129.10.131.166:8080/?action=stream
http://89.39.27.182:8080/?action=stream
http://66.130.223.168:8080/?action=stream
http://98.30.39.160:8080/?action=stream

BraINstinct0, 2020-12-02:
> security cams
> http://46.16.226.181:88/?action=stream
> http://79.132.232.174:8080/?action=stream
> http://46.151.101.158:8081/?action=stream
> http://90.66.111.157:8080/?action=stream [...]

where did you get those?

erikmartinjordan, 2021-01-13:
I couldn't find any working RTSP streams from the links above. Someone knows a reliable RTSP public link?

crossworth, 2021-01-14:
@erikmartinjordan 
rtsp://wowzaec2demo.streamlock.net/vod/mp4:BigBuckBunny_115k.mov

bobby20180331, 2021-01-29:
> http://68.3.132.170:8080/?action=stream
> http://129.10.131.166:8080/?action=stream
> http://89.39.27.182:8080/?action=stream
> http://66.130.223.168:8080/?action=stream
> http://98.30.39.160:8080/?action=stream

equipment cameras work！

tabishimran, 2021-01-29:
@bobby20180331 , yea most of these links stop working after a while, I wrote a crawler to index these links a while ago because I needed some links to test something I was building.

tomaszek92, 2021-11-16:
rtsp://rtsp.stream/movie
rtsp://rtsp.stream/pattern

rtspstream, 2022-01-09:
Thanks for posting these URLs! I am the creator of https://rtsp.stream and I put together this service after struggling to find a reliable RTSP Stream to test my software application. I would like to hear from the community on how you're using these URLs and if there's any feature that you'd like to see. Considering the server/bandwidth costs, I also plan to introduce a paid plan (while still maintaining the current features as part of the free plan). What are some of the features you'd be happy to pay for? Please reach out to me on dave@rtsp.stream with your feedback and suggestions, thank you!

fpaupier, 2022-02-21:
> Thanks for posting these URLs! I am the creator of https://rtsp.stream and I put together this service after struggling to find a reliable RTSP Stream to test my software application. I would like to hear from the community on how you're using these URLs and if there's any feature that you'd like to see. Considering the server/bandwidth costs, I also plan to introduce a paid plan (while still maintaining the current features as part of the free plan). What are some of the features you'd be happy to pay for? Please reach out to me on [dave@rtsp.stream](mailto:dave@rtsp.stream) with your feedback and suggestions, thank you!

Thanks a lot for this service Dave! I used your `rtsp` to test a videostream project (performing rtsp towards mp4 conversion) (https://github.com/horgh/videostreamer/pull/23)

hdanielmora, 2022-04-13:
> dave

hi Dave,  Thx a lot. I just started to play around with it. perhaps an authenticated version could be nice.

necrodome, 2022-12-21:
I am building a machine learning pipeline that involves RTSP streams and needed sample streams to test on. It's pretty easy to run your own public service with the help of [https://github.com/aler9/rtsp-simple-server](rtsp-simple-server). Here are some sample streams to test if you need.

- [rtsp://rtspstream.com/pattern](rtsp://rtspstream.com/pattern) 
- [rtsp://rtspstream.com/ball](rtsp://rtspstream.com/ball)
- [rtsp://rtspstream.com/parking](rtsp://rtspstream.com/parking)

<table>
  <tr>
    <td>rtsp://rtspstream.com/pattern</td>
     <td>rtsp://rtspstream.com/ball</td>
     <td>rtsp://rtspstream.com/parking</td>
  </tr>
  <tr>
    <td valign="top">[image]</td>
    <td valign="top">[image]</td>
    <td valign="top">[image]</td>
  </tr>
 </table>

The first two are GStreamer patterns with the current time and fly.io region overlayed on top.
The third one is a top-view parking video

While I can't guarantee reliability, you can run your own (for free, if you stay within the limits) on fly.io by following the instructions here: https://github.com/necrodome/rtsp-on-flyio

guytechNet, 2023-06-12:
Hi necrodome,
I can not use three rtsp as follows:
rtsp://rtspstream.com/pattern
rtsp://rtspstream.com/ball
rtsp://rtspstream.com/parking

This is message error
[image]

rtspstream, 2023-06-12:
@guytechNet try https://rtsp.stream for a hosted service

necrodome, 2023-06-12:
> Hi necrodome, I can not use three rtsp as follows: rtsp://rtspstream.com/pattern rtsp://rtspstream.com/ball rtsp://rtspstream.com/parking
> 
> This is message error [image]

I suggest running your own instance where you can control the whole pipeline using https://github.com/necrodome/rtsp-on-flyio .

USAMAWIZARD, 2023-07-05:
[Ant Media](http://antmedia.io/) An RTSP server which supports WebRTC , RTMP , HLS and Dash output.

pierro42, 2023-12-02:
Bump, this can help a lot of people debug their stream before thinking that the app is bugged.

AzwadFawadHasan, 2023-12-19:
anyone knows any software that will convert a private rtsp stream into a public/http so that I can view it through the web

USAMAWIZARD, 2023-12-19:
@AzwadFawadHasan , Yes its possible through [Ant Media](http://antmedia.io/), you can contact me at usama.tahseen@antmedia.io , would love to help you with this requirement.

AzwadFawadHasan, 2023-12-21:
thanks i'll check it out

SanDyLibs, 2024-02-28:
Try these URL's :

**_rtsp://rtspstream:021ce389961591228dee593b83439e76@zephyr.rtsp.stream/movie_**
**_rtsp://rtspstream:74ded9025fe6d0ff2d1f20ec26b55591@zephyr.rtsp.stream/pattern_**

rtspstream, 2024-02-29:
> Try these URL's :
> 
> **_rtsp://rtspstream:[021ce389961591228dee593b83439e76@zephyr.rtsp.stream](mailto:021ce389961591228dee593b83439e76@zephyr.rtsp.stream)/movie_** **_rtsp://rtspstream:[74ded9025fe6d0ff2d1f20ec26b55591@zephyr.rtsp.stream](mailto:74ded9025fe6d0ff2d1f20ec26b55591@zephyr.rtsp.stream)/pattern_**

Please DO NOT use those URLs directly. These are private URLs for the account and they have a limit of 2GB / Month of video transfer. Instead, create a free account on https://rtsp.stream and use your own set of RTSP URLs.

@SanDyLibs please delete these, they are not supposed to be made public and are specific to your account.

baerrus, 2024-03-04:
> Most of the above URLs are no longer active. The following it is a currently working one: **rtsp://rtsp-test-server.viomic.com:554/stream**

Not working

AzwadFawadHasan, 2024-03-05:
@baerrus Hi, you can check this repo. These contain some of the most stable free rtsp links out there

https://github.com/AzwadFawadHasan/Public_RTSP_Sources

AzwadFawadHasan, 2024-03-05:
I have curated some of the working rtsp stream. Feel free to checkout the repo below

https://github.com/AzwadFawadHasan/Public_RTSP_Sources

baerrus, 2024-03-05:
@AzwadFawadHasan none of these links have anything to do with RTSP. They are motionJpeg streams. What am I missing? Why do you call them RTSP?

AzwadFawadHasan, 2024-03-05:
@baerrus my bad. I'll soon fix it.

AzwadFawadHasan, 2024-03-05:
@baerrus  I had saved some rtsp streams before. Those don't work now.  Just checked the wowza demo rtsp stream, it also stopped working for me. I guess you could create a RTSP server or use a third party media server that has rtsp output plugins.

rocxteady, 2024-04-08:
> @baerrus my bad. I'll soon fix it.

You still didnt delete your comments about RTSP.

ToshibaMastr, 2024-07-19:
**rtsp://62.109.19.230:554/iloveyou**
Works only with tcp connection (ffplay -rtsp_transport tcp ...)

ToshibaMastr, 2024-08-07:
> **rtsp://62.109.19.230:554/iloveyou** Works only with tcp connection (ffplay -rtsp_transport tcp ...)

I've updated the server, now it won't die when I'm not watching it, and you also have the opportunity to launch your streams using the command:
```sh
gst-launch-1.0 file src location=file.mp4 ! qtdemux name=d d.video_0 ! queue ! rtspclientsink protocols=tcp+udp name=s location=rtsp://62.109.19.230:554/(name) d.audio_0 ! queue ! s.
```
or
```sh
ffmpeg -re -stream_loop -1 -i file.mp4 -c copy -f rtsp rtsp://62.109.19.230:554/(name)
```

estate-car, 2024-09-10:
I have a pretty important thing:
Make sure that VLC is allowed on your system's firewall

Gabka365, 2024-12-20:
Here are two probe streams for testing
https://rtsp.stream/auth/login

ShayanFiroozi, 2025-02-15:
A beautifual video found in **2025** !
[rtsp://807e9439d5ca.entrypoint.cloud.wowza.com:1935/app-rC94792j/068b9c9a_stream2](rtsp://807e9439d5ca.entrypoint.cloud.wowza.com:1935/app-rC94792j/068b9c9a_stream2)

tinncdev, 2025-03-15:
found an app that create either a dummy pattern or from camera, look like no ads as well.
saved me a lot of time checking all dead url above
https://apps.apple.com/vn/app/rtsp-stream/id6474928937

rtspstream, 2026-02-15:
> found an app that create either a dummy pattern or from camera, look like no ads as well. saved me a lot of time checking all dead url above https://apps.apple.com/vn/app/rtsp-stream/id6474928937

yup, our team built that app as well, so we now have both an iOS app ([OctoStream RTSP Server](https://apps.apple.com/au/app/octostream-rtsp-server/id6474928937)) as well as hosted RTSP streams on http://OctoStream.com. Both are free. The website has bandwidth caps to make sure the servers are not overwhelmed. Use whatever works best. And please let us know if you guys are looking for more features. We would be happy to add more support.

volkov-maxim, 2026-03-15:
Another way: create RTSP stream for test from Youtube video/live with [go2rtc](https://github.com/AlexxIT/go2rtc).
[In my case, I did it for Frigate NVR](https://share.google/aimode/LHjyP79knlvdbetJO).