grigory-lobkov/rtsp-camera-view #5: how to connect to multicast rtsp camera ?

BalarajAdappa, 2018-05-21:
Hi @grigory-lobkov  i'm using your stream viewer, thanks is not enough to say you, i can connect to unicast stream camera using your app, now i want to receive multiicast stream , what changes should i  do to get it done?? help would be appreciated.

grigory-lobkov, 2018-05-21:
Good day, @BalarajAdappa 

You can use rtsp-camera-view'er with any string, you provide to vlc Media -> Open Network Stream
[image]

You need to find the url string to connect your camera multicast port and try to open in VLC first.

The first instruction, I've found is for [Dahua NVR](https://dahuawiki.com/NVR/Recording_Setup/Multicast_Setup)

BalarajAdappa, 2018-05-22:
Hi @grigory-lobkov   i have tried it but i could not receive multicast stream . the unicast adrress id= rtsp://192.168.1.13:554/JPEG 90000 and multicast ip address is=rtsp://@224.0.0.10:554/JPEG 90000 when i connect to unicast connectoin will establish and data will be sent to multicast group, (i can see data in wireshark) but when i try to connect to multicast address using vlc it is not working

grigory-lobkov, 2018-05-22:
Unicast is used to decrease camera CPU load, when a lot of clients connected. I don't know about others, but it seems, your camera does not generate multicast until unicast is used by somebody. Why don't you use unicast, if you are only one observer? ...

BalarajAdappa, 2018-05-26:
Hi @grigory-lobkov  i have tried unicast its working fine, i want multiple observer to see my stream i want to try for multicast.

grigory-lobkov, 2018-05-26:
1)
Maybe, you have one client, you shure, he will be always connected - just give him a unicast address, multicast for others
2)
Search in camera web-console for option "multicst always"
3)
Ask camera vendor for "multicst always" ability

BalarajAdappa, 2018-06-02:
Hi @grigory-lobkov  May i know this application in available for android platform like Android TV or Phone??

Thanks .

grigory-lobkov, 2018-06-03:
I have seen a lot of programs on google play ([rtsp view](https://play.google.com/store/search?q=rtsp%20view&c=apps), [ip camera view](https://play.google.com/store/search?q=ip%20camera%20view&c=apps)), try it. I've used only programs for one camera.
Do you have NVR?
You can ask NVR-vendor for Android software to connect it.

I'm also using free [ONVIFER](https://play.google.com/store/apps/details?id=net.biyee.onvifer) for one-camera view and PTZ-control. But on the pictures I've found multi-camera view. You can try this also.

grigory-lobkov, 2018-06-18:
Please, check out a new [beta 1.0.0.5](https://github.com/grigory-lobkov/rtsp-camera-view/releases/tag/1.0.0.5) !

closed 2018-06-18 by grigory-lobkov

BalarajAdappa, 2018-06-19:
@grigory-lobkov thanks and is it possible to dump stream into pc or external disk? if possible please guide me.

thanks regards
Balaraj adappa

grigory-lobkov, 2018-06-19:
For now I only think about necessity of this function, because this product called "camera view", but not record, not register. And it's quite difficult to do it good and conveniently.

grigory-lobkov, 2018-06-19:
You can try http://www.ispyconnect.com/download-agent.aspx
and https://iss.ru/upload/securos-lite-9.6-R1.zip (maybe only russian language)
for this purposes

BalarajAdappa, 2018-06-20:
Thank you @grigory-lobkov  i will check them  and writr you my opinion.