grigory-lobkov/rtsp-camera-view #7: Crash after running for a short time (Beta 1.0.0.4)

HayStax, 2018-05-29:
I have had the beta version crash after running for maybe about 10 minutes twice now the last crash had this info,

See the end of this message for details on invoking 
just-in-time (JIT) debugging instead of this dialog box.

************** Exception Text **************
System.NullReferenceException: Object reference not set to an instance of an object.
   at Presenter.Presenters.SourcePresenter.StopOnInvisible()
   at View.Components.SourceControl.Invoke(Action action)
   at View.Components.SourceControl.<.ctor>b__43_8(Object sender, EventArgs args)
   at System.Windows.Forms.Timer.OnTick(EventArgs e)
   at System.Windows.Forms.Timer.TimerNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

************** Loaded Assemblies **************
mscorlib
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.2650.0 built by: NET471REL1LAST_B
    CodeBase: file:///C:/Windows/Microsoft.NET/Framework/v4.0.30319/mscorlib.dll
----------------------------------------
RtspCameraView
    Assembly Version: 1.0.0.4
    Win32 Version: 1.0.0.4
    CodeBase: file:///C:/Data/Apps/rtsp-camera-view.1.0.0.4.beta/RtspCameraView.exe
----------------------------------------
System.Windows.Forms
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.2634.0 built by: NET471REL1LAST_C
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms/v4.0_4.0.0.0__b77a5c561934e089/System.Windows.Forms.dll
----------------------------------------
System
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.2634.0 built by: NET471REL1LAST_C
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System/v4.0_4.0.0.0__b77a5c561934e089/System.dll
----------------------------------------
System.Drawing
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.2634.0 built by: NET471REL1LAST_C
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Drawing/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Drawing.dll
----------------------------------------
Presenter
    Assembly Version: 1.0.0.0
    Win32 Version: 1.0.0.0
    CodeBase: file:///C:/Data/Apps/rtsp-camera-view.1.0.0.4.beta/Presenter.DLL
----------------------------------------
System.Core
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.2634.0 built by: NET471REL1LAST_C
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Core/v4.0_4.0.0.0__b77a5c561934e089/System.Core.dll
----------------------------------------
ViewVlc215
    Assembly Version: 1.0.0.0
    Win32 Version: 1.0.0.0
    CodeBase: file:///C:/Data/Apps/rtsp-camera-view.1.0.0.4.beta/ViewVlc215.DLL
----------------------------------------
Model
    Assembly Version: 1.0.0.0
    Win32 Version: 1.0.0.0
    CodeBase: file:///C:/Data/Apps/rtsp-camera-view.1.0.0.4.beta/Model.DLL
----------------------------------------
LightInject
    Assembly Version: 3.0.1.9
    Win32 Version: 3.0.1.9
    CodeBase: file:///C:/Data/Apps/rtsp-camera-view.1.0.0.4.beta/LightInject.DLL
----------------------------------------
System.Security
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.2650.0 built by: NET471REL1LAST_B
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Security/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Security.dll
----------------------------------------
Accessibility
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.2053.0 built by: NET47REL1
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/Accessibility/v4.0_4.0.0.0__b03f5f7f11d50a3a/Accessibility.dll
----------------------------------------
System.Configuration
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.2053.0 built by: NET47REL1
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Configuration/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Configuration.dll
----------------------------------------
System.Xml
    Assembly Version: 4.0.0.0
    Win32 Version: 4.7.2634.0 built by: NET471REL1LAST_C
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Xml/v4.0_4.0.0.0__b77a5c561934e089/System.Xml.dll
----------------------------------------
Microsoft.GeneratedCode
    Assembly Version: 1.0.0.0
    Win32 Version: 4.7.2634.0 built by: NET471REL1LAST_C
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Xml/v4.0_4.0.0.0__b77a5c561934e089/System.Xml.dll
----------------------------------------
AxInterop.AXVLC
    Assembly Version: 1.0.0.0
    Win32 Version: 1.0.0.0
    CodeBase: file:///C:/Data/Apps/rtsp-camera-view.1.0.0.4.beta/AxInterop.AXVLC.DLL
----------------------------------------
Interop.AXVLC
    Assembly Version: 1.0.0.0
    Win32 Version: 1.0.0.0
    CodeBase: file:///C:/Data/Apps/rtsp-camera-view.1.0.0.4.beta/Interop.AXVLC.DLL
----------------------------------------

************** JIT Debugging **************
To enable just-in-time (JIT) debugging, the .config file for this
application or computer (machine.config) must have the
jitDebugging value set in the system.windows.forms section.
The application must also be compiled with debugging
enabled.

For example:

<configuration>
    <system.windows.forms jitDebugging="true" />
</configuration>

When JIT debugging is enabled, any unhandled exception
will be sent to the JIT debugger registered on the computer
rather than be handled by this dialog box.

grigory-lobkov, 2018-05-30:
Thank's, you testing my app!
It seems, you maximize one stream, and others became invisible.
If so, after a while invisible streams are stopped to save bandwidth. And some error happen.
I will try to find and fix it. If this error disturbs you, you can change matrix size to 1x1.

grigory-lobkov, 2018-06-18:
Please, check out this issue in new [beta 1.0.0.5](https://github.com/grigory-lobkov/rtsp-camera-view/releases/tag/1.0.0.5)

grigory-lobkov, 2018-07-16:
[1.0.0.6](https://github.com/grigory-lobkov/rtsp-camera-view/releases) released.
Please, say you opinion.