grigory-lobkov/rtsp-camera-view #8: Grey Tags showing after trying the latest beta

IraqiNinja, 2018-06-12:
[image]

This keeps showing on the new Beta version, any way to get rid of it?
{name} LOST link
{name} recovered link

Working amazing other than that.

Thanks a lot.

grigory-lobkov, 2018-06-12:
Thank you for testing my program!
I will fix it in new beta!

Sorry, for such a blunder!

You give me the strength to go forward.

IraqiNinja, 2018-06-12:
Any way i can donate to your project, I am new here, sorry if its a dumb question :)

grigory-lobkov, 2018-06-18:
Please, check out this issue in new [beta 1.0.0.5](https://github.com/grigory-lobkov/rtsp-camera-view/releases/tag/1.0.0.5)

grigory-lobkov, 2018-07-16:
[1.0.0.6](https://github.com/grigory-lobkov/rtsp-camera-view/releases/tag/1.0.0.5) released!
Waiting for your opinion...

closed 2018-07-22 by grigory-lobkov