grigory-lobkov/rtsp-camera-view #9: More then 8 cameras?

Yeomans, 2018-06-26:
Hi,

Is there an issue that prevents more than 8 cameras being connected?

It seem that when a 9th is connected it just shows the VLC icon on a white background.

Cheers

Yeomans, 2018-06-26:
Actually nevermind...seems that restarting the program after configuring allowed it to work!

Yeomans, 2018-06-26:
Ok I spoke too soon haha. It seems like going more than 9 makes the program unstable and it crashes

grigory-lobkov, 2018-06-26:
Thanks for testing my app!
I'm trying to do my best to make app able working 24/7/365.
Your case is really interesting!

Please, write an OS version (with SP#), Framework version, my app version. I will test it.
As I understood, to reproduce the bug, I need to do:
1) delete config (or install fresh 1.0.0.5)
2) when default matrix 2x2, adding 10 cameras to list
3) add 4 cameras to preview
4) do matrix size 3x3 (on this step I will get one white  rectangle)
5) add 5 more cameras to preview
6) do matrix size 4x4
7) add 1 more camera to preview (on this step the app will crash in 1-10 min)

Is it right?

Can you check the CPU load for each core & Ethernet bandwidth, when you using the app?
I will try to make screen capture (by bandicam) when testing. If you able, please, make it too.

Ziowebbo, 2018-06-27:
Good morning, I'm using a matrix of 3x3 without any problem. In my case I also use 3 instances of the software, each of 3x3 matrix for a total of 27 streams on a single computer.
May I suggest a lack of power by the processor or the graphics unit?

grigory-lobkov, 2018-06-27:
My Win10 i3-6300 can handle only 13 threads of 960p 15fps (some of them is 1080p).
If I add one more camera, one CPU core load became 110% and VLC can't start a new stream.
[screen capture](https://cloud.mail.ru/public/2aN1/EB5ffQuKA)

But first time I've started to add cameras, some strangenesses happens when adding 10-12 cameras.
I think Windows learned, what resources need to my app...
[screen capture](https://cloud.mail.ru/public/5sq4/CBUea7yJW)

When I switched to substreams (640x480 20fps), I got only 8 additional cameras. Not much.
[screen capture](https://cloud.mail.ru/public/JdPz/aJwd7ou76)

**To beat this issue, I need to split sources to separate threads (asking help in [#2](https://github.com/grigory-lobkov/rtsp-camera-view/issues/2))**

My advice will be to use a high GHz processors, and **minimal substream resolution**, plus minimal fps. When accident occurs, double click will switch camera for good-resolution stream.

May be a good cheap solution will be in using **several monitors**, connected to one PC, as @Ziowebbo advice. Then, the CPU load will split between CPU cores.

Yeomans, 2018-06-28:
Interesting.

Yea i'm not too sure why i was having issues configuring it the first time around as they didn't come back on subsequent setups. Perhaps just something else on my computer at the time was affect int. Feel free to close this issue as the reason I opened it is no longer relevant.

grigory-lobkov, 2018-06-29:
No problem.
Please, tell me if the program has met your needs, or not.
You can do it in this issue, or by e-mail gg81 -at- yandex -dot- ru

Thank you!

closed 2018-06-29 by grigory-lobkov