grigory-lobkov/rtsp-camera-view #10: Person

Ziowebbo, 2018-06-27:
Hi, 
it would be great if the software had the possibility to resize the Video in the matrix to create (if user need it) a camera view in different dimension. This will allow to have a focus on a camera and have others smaller.
Try to add an image for example.
[image]

grigory-lobkov, 2018-06-27:
Thank you for this idea, I will try to do it in summer.
The other big enhancement is to move to VLC 3... But I think, I need to release something first :)

grigory-lobkov, 2018-07-16:
Please, test it in 1.0.0.6
https://github.com/grigory-lobkov/rtsp-camera-view/releases
I think, this will be a release candidate.

Ziowebbo, 2018-07-17:
Is working GREAT and flawlessy Grigory!!!!! 
Just a little thing: The TEXT, when you Join the "cells" the video is bigger and also the text, but risk to bee too large in respect of the other video.
Is possible to have a Fixed dimension only for the text?

grigory-lobkov, 2018-07-17:
Thanks. Done. Re-download the 1.0.0.6 :)

Ziowebbo, 2018-07-18:
Did it and is perfect, thanks!!! Gonn write you about the mail alert...