grigory-lobkov/rtsp-camera-view #11: Auto-Reconnect after signal loss

Ziowebbo, 2018-06-27:
Grigory, while we're at it I suggest a feature that we already talked about: auto-reconnect if a stream was lost.
thanks!

grigory-lobkov, 2018-06-27:
Thank's for testing my App, Zio!
Auto reconnect was implemented in 1.0.0.4 (and later). Does it work for you?

The reconnect timeout
- when first-connect/reconnect: 30s
- when already streaming: 10s

Please, try it. If it does not, I will test it too.

Ziowebbo, 2018-06-27:
Great, didn’t know that! 
I will try the feature!

grigory-lobkov, 2018-07-16:
[1.0.0.6](https://github.com/grigory-lobkov/rtsp-camera-view/releases/tag/1.0.0.5) released!
How does it work?

Ziowebbo, 2018-07-17:
Seems good, I'm trying to create a loss of connection to one of the video stream... but most important, I'm finalizing the SMTP configuration for the email alert!

grigory-lobkov, 2018-07-17:
Great! Closing...

closed 2018-07-17 by grigory-lobkov