grigory-lobkov/rtsp-camera-view #12: installation problem.

BalarajAdappa, 2018-06-29:
Hi @grigory-lobkov  when i install latest beta version its showing me error that "install vlc 2.1.5 " i followed insstructions and installed vlc then i tried to install your rtsp player still getting same [error.]
(url
[image]
)

grigory-lobkov, 2018-06-29:
Thank you for using my app!
It's a pitty, you met error before use :(

Please, reinstall VLC:
1) install it to default folder;
2) be sure, "ActiveX Web-module" is selected.

In beta version I tried to make errors more detailed.

BalarajAdappa, 2018-07-05:
Hi @grigory-lobkov  i have tried both vlc 2.1.3 and vlc 2.1.5 also.

grigory-lobkov, 2018-07-05:
Please, try to use installation version (not portable) of Release.
Try to get, what additional products it will install.
May be it will be "Microsoft Visual C++ Redistributable", or something like this.
Or "Microsoft framework".

And say me, what installed/happened, please...
You can take a screenshots, if possible...

grigory-lobkov, 2018-07-16:
@BalarajAdappa how does it going?
I've released [1.0.0.6](https://github.com/grigory-lobkov/rtsp-camera-view/releases/tag/1.0.0.5)
Please, check it out!

closed 2018-07-22 by grigory-lobkov