grigory-lobkov/rtsp-camera-view #13: Customizing text of Mail Alert

Ziowebbo, 2018-07-18:
As object says: do you think is possible to customize the email that the app send?

Thanks

Ziowebbo, 2018-07-18:
> Please, propose the text, you want.

I was thinking to a free editing text box with special “items” to use channel name, events ecc.
In this way everyone could use the language that wants.

grigory-lobkov, 2018-07-18:
You can edit your language by copying from "en.txt". You can customize the mail message, if you want. I don't see first need in this. The message is exists. It's informative.

No need to decorate it much.
BTW, I can add your language file to the project.

Ziowebbo, 2018-07-18:
Is perfect for me to edit the en.txt file mate!

Didn’t know it was possible!

Inviato da Iphone

grigory-lobkov, 2018-07-24:
If you do fully translated it.txt from en.txt, please, give it to me - I will upload it )

Ziowebbo, 2018-07-24:
Yes mate, for sure! But maybe I not understand, where I can change the mail text?

Inviato da Iphone

grigory-lobkov, 2018-07-24:
Is your OS shows you Italian messages?
1) copy en.txt to it.txt
2) edit "it.txt" and translate each line after "=" sign (and comment?)
3) open program and check strings

grigory-lobkov, 2018-07-29:
Have you found "en.txt" file ?
Have you tried to find part of text of eMail message in there ?
P.S. can you find
"SourceGrid/EmailOnLostSignalSubject = "

Have copied en.txt to it.txt ?

Ziowebbo, 2018-08-02:
Yes mate, I found it and I'm working on the translation, thanks!
I don't have much time during those days... everybody are on vacation and
I'm practically alone at work.

Sorry for delay in answering

Ziowebbo, 2018-08-07:
Ok mate, all is working!!!!
Just another question: if I would like to send the email alert to more than
one recipient, how I add to write and divide the addresses?
Something like "aaa@ccc.com;bbb@ccc.com" should work?

Thanks

2018-08-02 16:59 GMT+02:00 Zio Web <ziowebbo@gmail.com>:

grigory-lobkov, 2018-08-09:
It was so quiet here, I didn't checked for new messages :)

Translation:
When you will finish, give your language file to me, please, I will add it :)

Several recipients:
Make a new Issue about it. I will add it to next beta.

Ziowebbo, 2018-08-10:
Ok mate, sorry to broke the peace :D 

Translation is in preparation!

Have a nice day!

Inviato da Iphone

grigory-lobkov, 2018-08-13:
No problem, it was a joke, because my answer was long too.

closed 2018-09-30 by grigory-lobkov