grigory-lobkov/rtsp-camera-view #21: Change camera in grid from one to another .

BalarajAdappa, 2018-09-18:
Hi @aztazt @grigory-lobkov  is it possible to change my streaming camera from on to another in grid just changing tittle of camera i mean name of camera?? i tried doing it myself but i failed to do that.

grigory-lobkov, 2018-09-23:
> changing tittle of camera

Explain please...
Do you mean, that by clicking on title TextEditBox must appear. If you change/edit title to existing in camera list, stream of camera should change to existing camera...

> tried doing it myself 

How did you tried this? Downloaded source and tried to edit camera name component?

You can do like this: create matrix, containing all your cmeras. Make all your sources visible. Then you can choose any of your sources by double click on it. Or you can do one source bigger than others and then by dragging move, switch between big and desired cameras.

BalarajAdappa, 2018-09-25:
@grigory-lobkov "Hi what i meant is suppose i have 50 RTSP streaming cameras in my network, instead of modifying camera name &  RTSP path every time for 12 grid , Assign name and RTSP path to to all cameras, and just list 12 at time by changing name of camera" i tried doing it felt diffucult so i asked.