grigory-lobkov/rtsp-camera-view #22: Help regarding RTSP Connection Re-Establishment when there are interrupts.

BalarajAdappa, 2018-09-25:
Hi @grigory-lobkov  i wanted help regarding connection re-establishment, 
scenario1- streaming is happening , suppose there is network loss for about few minutes our stream player will loss connection and then network works normal then Player should Re-establish connection to stream server ,
scenario2-someone accidentally  unplug network cable , and plug back then player should Re-establish connection.  

please suggest what is the improvements we can do.

grigory-lobkov, 2018-09-30:
please, check this out, it should already work.

dinesh1218, 2019-02-14:
@grigory-lobkov I'm also in need of same scenario, While using RTSP, the connection needs to be re-established when the network is recovered from failure. Can you please help me with this?

grigory-lobkov, 2019-02-14:
It's done already!
Just wait a little (up to 10 minutes), and connection will reestablish!

P.S. If you tried and waited some time, and the camera was not recovered, then, say, what are you doing step-by-step.

dinesh1218, 2019-02-14:
@grigory-lobkov Thanks for your reply. It's automatically reconnects once is network is back.

grigory-lobkov, 2019-02-14:
And if network will disconnect several times, it's not reconnected?

dinesh1218, 2019-02-15:
@grigory-lobkov Yeah, I tried disconnecting the network a couple of times, it's getting reconnected.