grigory-lobkov/rtsp-camera-view #24: Show options

Reddds, 2019-03-14:
Было бы хорошо добавить настройки показа поверх остальных окон (чтобы панель задач не уменьшала видимую область) и настройку, скрывающую заголовок окна.

It would be nice to add settings: display on top of other windows (so that the taskbar does not reduce the visible area) and settings that hide the window title.

grigory-lobkov, 2019-03-15:
Пожалуйста, попробуйте параметр командной строки fullscreen=1 / fullscreen
В контекстом меню есть опция "На весь экран". Может быть это подойдет?...

Please, try command line parameter "fullscreen=1" / "fullscreen"
There's option in context menu "Full screen". Does it suits you?...

grigory-lobkov, 2019-05-21:
Can I close it ?
Does it suits you ?

closed 2019-07-14 by grigory-lobkov