grigory-lobkov/rtsp-camera-view #27: Not a problem, just recomendation

zxcslo, 2019-04-30:
I like this program very much. But I have a recommendation...
It would be very useful if the program would remember the size and the position when you close it. So next time it would open at the same size and position. Now you must resize it and position it every time...
Could this be done, please?

grigory-lobkov, 2019-05-21:
What is the purpose of different position and size, you ask for ?
For now program have ability to set screen and full size in command line parameters: [https://user-images.githubusercontent.com/36440722/36472447-aaf96b04-1713-11e8-921c-6fa95168e6a7.jpg](url)

zxcslo, 2019-05-21:
I have 6 cameras and I like to view them on my 4k monitor at the top of the screen. I resize the window to this: [https://drive.google.com/file/d/1SeOp9ukTJvFWRnYBDGGXvOZ6JzaZGgPO/view?usp=drivesdk](https://drive.google.com/file/d/1SeOp9ukTJvFWRnYBDGGXvOZ6JzaZGgPO/view?usp=drivesdk).
But when I close the program and reopen it then it's like this: [https://drive.google.com/file/d/1EQMZBPJrsdTtCK4uz5BlHGOT0CPgHTvB/view?usp=drivesdk](https://drive.google.com/file/d/1EQMZBPJrsdTtCK4uz5BlHGOT0CPgHTvB/view?usp=drivesdk)

grigory-lobkov, 2019-05-22:
I see... I think, It's a good function.
Will try to find some time to add it.

zxcslo, 2019-12-31:
Happy new year.