grigory-lobkov/rtsp-camera-view #28: using tcp instead of udp rtsp stream

da-sti, 2019-05-10:
Hi !

just discovered this nice project, very clean interface

I have one problem with fw cameras that are beyond a vpn and I can't figure out for now how to force them to send udp packet smaller the 1300 but it works with TCP and vlc works also with the setting you suggest here https://github.com/grigory-lobkov/rtsp-camera-view/wiki/VLC-is-showing-ghosts-on-RTSP-processing

I sniffed the network and when I use RTSP camera view, it uses UDP whatever I configure VLC to use udp or tcp.

Is it possible for your software to use the settings of VLC ?

Another question, is it possible to use the last version of VLC ?

thanks for this great project
david

grigory-lobkov, 2019-05-22:
Hello!
Sorry, I don't have much time for supporting this project for now.
It doesn't feed's me :)
Have you tried to disable udp on camera?

Can't help your for now. Maybe someone from community will help to add such a functionality.

da-sti, 2019-05-22:
Hi !

the core problem is not in rtsp camera, it's a network one : when cameras streams in UDP, the packets are too big and they will fall into a network traffic black hole (certainly caused by VPN and/or ISP or something else I didn't find yet). Whatever I do, I can't get them send smaller packets for now.

The workaround I found with VLC, is to be in [TCP mode] (like in the wiki) but RTSP_camera doesn't "follow" VLC's setting and keeps getting the stream in UDP

Anyway, thanks for this project, the software is already very nice as it is (easy to use, fast, use existing standards...).
I hope the community will help.

RuStrannik, 2019-10-02:
Exactly the same problem for me. The program simply ignores the recommended setting "RTP over RTSP (TCP)", no matter what I do, and simply gets stuck after the first frame even on the local network...
Gregory, since this setting is in the FAQ, could you please tell how does your program suppose to follow this setting? Does check for "rtsp-tcp=1" in the "vlcrc" file, or simply relies on vlc to pickup the config somehow?

grigory-lobkov, 2019-10-03:
I'm sorry, hearing that.
This settings in FAQ for fighting with ghosts, but not with protocol.
I think, the protocol have to be set up in your cameras. Also, you can try to forward only TCP on your router, when camera is over NAT.
Program doesn't take this settings, VLC using it itself.