grigory-lobkov/rtsp-camera-view #30: any way to do this?

Graham0264, 2019-07-19:
I'm trying to use this on W10 with dual monitors. What I'd really like is to have a matrix of camera views on one monitor with a full screen view on the other. Click on a view on the matrix and it replaces the full screen one.
Any way to do this?

shwnchr, 2019-08-18:
I don't know how to achieve what you are asking but I've thought of a workaround if you're willing to compromise. Details follow.

Run two instances of the program, with identical matrices, one on each monitor. Double-click the camera that you want displayed full-screen on the monitor where you want it displayed full-screen. Now you have the matrix displayed on one monitor and the full-screen camera displayed on the other.

I can think of several reasons why this arrangement is less than ideal but for lack of a better option?

shwnchr, 2019-08-18:
Here's another solution but it comes with a caveat: I don't have a multi-mon setup to try this on, so YMMV. Also, the description below assumes a 3x3 grid of 9 cameras, modify as needed. Here goes...

**ABSTRACT**
Change your 3x3 matrix to a 6x3 matrix with the 3 extra columns joined. The entire matrix would then be displayed as an overflow onto the second monitor.

**STEP-BY-STEP**

**1.** *Click [Settings], [Matrix dimensions]*
[image]

**2.**  *Click and drag to select, as shown*
[image]

**3.** *Click [Join]*
[image]

**4.** *Click [Save]*
[image]

**5.** *Drag the desired camera from the 3x3 grid into the 1x1 grid.*
[image]

**6.** Here's the part I can't replicate for you. If necessary, un-maximize the window and manually stretch it across until it completely covers both monitors.
[image]