grigory-lobkov/rtsp-camera-view #35: VLC can not be found by program on Windows 10

yxm0513, 2019-12-25:
VLC was installed with default path without any issue, and then run the program, still get an error as :

[image]

bisewski001, 2019-12-30:
Not just in 10, but in Seven too.

gomezpt, 2020-01-04:
No compatibility with VLC 3.x.x ?

RuStrannik, 2020-01-04:
Just in case it might work for you:
https://stackoverflow.com/questions/59180160/mosaic-of-rtsp-streams-vlc-3-0-8

grigory-lobkov, 2020-04-25:
Hi, guys.
You have to install VLC 2.1.5 with "Web plugins" option in installer selected.
[image]

There's no compatibility with vlc 3.x for now, sorry.

closed 2020-04-25 by grigory-lobkov