grigory-lobkov/rtsp-camera-view #36: Why Not Always Expected Switching Behavior?

makoven, 2020-01-23:
Hi and thanks for this project!

I added a couple of cameras to the grid. Double-clicked on one of them to expand to full screen. The camera switched to a stream of good quality. Then doble-clicked again. And I noticed that the camera is still in good quality and does not return to the secondary stream even on a grid

Is it possible to get the expected behavior always: good quality of one camera if it is full screen. And the secondary stream for ALL cameras when they are not full screen grid?

fragtion, 2020-04-22:
I've done some testing with source code, and found the cause for this problem.

Every few seconds, each camera feed is tested for Width & Height Value of the video feed, vs Width & Height of the viewer/player dimensions which it is in. If the viewer height & width is larger than the video height & width, it assumes you are in full screen mode

Sometimes the feed dimensions are larger than viewer dimensions for that video even if you are not in full screen mode but matrix mode, so the feed switches to (or remains on) HQ feed.  I am investigating for possible fix while author is away. If I find a good workaround I will advise here

makoven, 2020-04-23:
@fragtion 
I'm not in a hurry. Let's hope the author accepts your corrections. Thanks for participating

grigory-lobkov, 2020-04-25:
added in [1.0.3](https://github.com/grigory-lobkov/rtsp-camera-view/releases/tag/v1.0.3)

closed 2020-04-25 by grigory-lobkov

fragtion, 2020-04-25:
Greetings @grigory-lobkov 
I understand you are very busy! Thanks for the new release and bugfix 👍 
Please note that git repo source was not updated to reflect the changes yet, and also source zip for v1.0.3 on the Releases page seems to be the source code for v1.0.2 (no changes?). Please will you push source changes to git when you have a moment:)
Many thanks & Regards

grigory-lobkov, 2020-04-25:
Sorry. Updated. Should I re-release it ?
You can pull your requests - no problem)
I thought a lot about this new function, how to do it: in camera or in program options. I have no such options window for now, and there is no other options to add there.

Of cause, this option will make user to think, that's why it's better to move it somewhere. Maybe it would be better to leave it only in config file...