grigory-lobkov/rtsp-camera-view #37: New versions of VLC

rschwarzjr, 2020-02-26:
Is it possible to run this on current releases of VLC? I assume there is some major change preventing this.

fragtion, 2020-04-15:
Also need support for VLC3.x, hoping to see update soon !

mfkl, 2020-05-05:
I'd strongly encourage you guys to move to VLC 3.x, VLC 2.x has known security issues FYI.

VerodinDev, 2024-06-19:
We're halfway through 2024 and since this issue is still open, I'm commenting here so it might get some renewed attention. 

H265 has become widely used by now (e.g. even the cheapest camera modules come with it by default). Not to mention security fixes and overal decoding optimisations/improvements.

bclouthier, 2024-09-29:
I would hope we could get support for vlc 3.x and h.265 streams as well. This program seems perfect for my needs.