grigory-lobkov/rtsp-camera-view #39: При переключении экрана с большого на малый, поток не переключается на плохое качество

gthb3, 2020-07-30:
После переключения на одну из камер, включается режим хорошего качества, но при возвращении камеры в стандартный режим, качество этой камеры не меняется обратно, на плохое, такой функции очень хватает
--
After switching to one of the cameras, a good quality mode is turned on, but when the camera returns to standard mode, the quality of this camera does not change back to bad, this function is very enough

grigory-lobkov, 2020-07-30:
Запишите, плиз, видео записи с экрана и выложите в облако. Это переключение происходит, и если плохой поток ещё запрашивается с камеры, то происходит сразу через 1 сек. Если хорошее качество показывалось долго (~30 сек, то плохой поток обрывается), то потребуется немного времени, чтобы запросить с камеры плохой поток, и по готовности произойдет переключение.
Я во время тестов ориентировался на время, которое рисует камера на картинке - при плохой картинке и при хорошей высота времени в пикселах была одинаковая, а относительно общей картинки было заметно, когда происходит переключение. Как вы понимаете, что переключение не происходит?

Еще может быть вариант, что когда камера переключается на хороший поток, плохой она уже отдавать этому же клиенту не хочет - поэкспериментируйте с VLC.

gthb3, 2020-07-30:
Не ожидал столь скорого ответа
Видео к сожалению записать не могу, но ваш ответ меня полностью устраивает, видимо действительно какие-то особенности работы vlc
Проблему решил созданием нескольких профилей для каждой камеры с разными настройками разрешения

Не знаю, будет ли вам это как-то полезно, но строки для подключения к камерам были такие
плохое качество rtsp://172.16.1.1:554/axis-media/media.amp?videocodec=h264&resolution=640x480
хорошее rtsp://172.16.1.1:554/axis-media/media.amp?videocodec=h264&resolution=1280x720
Момент переключения проверял по нагрузке на сеть, через диспетчер задач win10

В связи с этим, возникла идея
При переключении профилей камер, даже если камера одна и та же, но с разными настройками, переключение всегда происходит хорошо, может быть можно реализовать переключение качества по такому же принципу?
Или, даже может быть, добавить кнопку с подобной функцией смены качества на панель рядом с громкостью?

Не сочтите за наглость, меня вполне устраивает программа в том виде, котором она есть сейчас, в любом случае, спасибо за прекрасную программку, ничего подобного, а тем более бесплатного в сети не находится

gthb3, 2020-08-03:
Проблема решена

closed 2020-08-03 by gthb3